raphaelbs/esp32-cam-ai-thinker #2: Have you solved the square problem?

9crk, 2019-01-16:
I'v tried the ESP32-CAM-DEMO stuff and just got a bad image. far from the youtube video did.
some times even have no image.and just a square.
have you solved this problem? can you take a good picture now?

9crk, 2019-01-16:
I saw you use a psram in your project. while I dont have one on my board. does it necessary?

raphaelbs, 2019-01-17:
I've managed to take UXGA pics (which is the higher resolution) with the code available here.
Unfortunately higher resolutions takes up quite a lot RAM space thus require external chips.
I'm not sure what's the biggest resolution you can get without the external RAM.

What about a example that profiles resolutions from the bottom up?