raphaelbs/esp32-cam-ai-thinker #3: Flashing led when using sdcard?

dflt00, 2019-02-06:
Hi, thanks for the example of writing jpg to the sd card, I couldn't find another example of this for the ESP32-cam anywhere. 

I have it working, but I have a question: does the flash led on your ESP32-cam also light up when using the sd slot? It seems GPIO4 is both connected to the flash led and to data1 of the sd slot. (Gives new meaning to the term 'flashing from SD'). But I thought maybe I'm missing something? 

(Of course not an issue with your fine example, but with the ESP32-cam hardware design)

raphaelbs, 2019-02-07:
Hey @dflt00,
Yep, you are right. When data is being transfer to or read from SD, the LED will blink. Not sure if we can bypass this without a hardware change.
Fortunately we can use both at different timestamps if needed.

dflt00, 2019-02-08:
Thanks for the confirmation. Well, other modules of this type don't have a flash led to begin with, and I can always remove it, so I don't consider it a drawback for this cheap module.

closed 2019-02-23 by raphaelbs

peterashwoodsmith, 2023-01-25:
There is a fairly easy fix to the hardware to this problem. Basically you decouple the LED flash from the SD card data line and splce it into the smaller LED GPIO pin. Description below.

https://www.youtube.com/watch?v=loet1ndG8Sg