raphaelbs/esp32-cam-ai-thinker #7: Porting to PlatformIO

EmmanuelKinyanjui, 2019-02-22:
Greetings,
Is it possible to port the existing code base to PIO using ESP-IDF?

raphaelbs, 2019-02-23:
It is possible, yes.
Though it require changes in the project structure, [as shown here](https://github.com/platformio/platform-espressif32/tree/master/examples/espidf-hello-world).

Why did you prefer PlatformIO over Make?

EmmanuelKinyanjui, 2019-02-25:
I have worked with PlatformIO before hence the preference. Will attempt to run it on PlatformIO and revert.

EmmanuelKinyanjui, 2019-02-25:
# Attempts to run it on PIO brings about the following error. 

--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5788
load:0x40078000,len:7900
ho 0 tail 12 room 4
load:0x40080000,len:5836
entry 0x40080314
␛[0;32mI (30) boot: ESP-IDF 3.30102.190109 2nd stage bootloader␛[0m
␛[0;32mI (30) boot: compile time 14:04:00␛[0m
␛[0;32mI (30) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (35) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (40) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (43) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (48) boot: Partition Table:␛[0m
␛[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (81) boot: End of partition table␛[0m
␛[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08b8c ( 35724) map␛[0m
␛[0;32mI (106) esp_image: segment 1: paddr=0x00018bb4 vaddr=0x3ffc0000 size=0x022c8 (  8904) load␛[0m
␛[0;32mI (110) esp_image: segment 2: paddr=0x0001ae84 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (114) esp_image: segment 3: paddr=0x0001b28c vaddr=0x40080400 size=0x04d84 ( 19844) load␛[0m
␛[0;32mI (131) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x17318 ( 95000) map␛[0m
␛[0;32mI (165) esp_image: segment 5: paddr=0x00037338 vaddr=0x40085184 size=0x036d8 ( 14040) load␛[0m
␛[0;32mI (176) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (176) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (176) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (180) cpu_start: Starting app cpu, entry point is 0x400817d8␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (191) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (197) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (203) heap_init: At 3FFC33A8 len 0001CC58 (115 KiB): DRAM␛[0m
␛[0;32mI (210) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (216) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (222) heap_init: At 4008885C len 000177A4 (93 KiB): IRAM␛[0m
␛[0;32mI (229) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (247) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (274) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (354) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (354) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (354) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (364) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (374) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (384) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (394) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (404) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (414) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (424) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (434) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (444) camera: Allocating 1 frame buffers (234 KB total)␛[0m
␛[0;32mI (444) camera: Allocating 234 KB frame buffer in PSRAM␛[0m
␛[0;31mE (454) camera: Allocating 234 KB frame buffer Failed␛[0m
␛[0;31mE (464) camera: Failed to allocate frame buffer␛[0m
␛[0;31mE (464) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first␛[0m
␛[0;31mE (474) camera: Camera init failed with error 0x101␛[0m
␛[0;31mE (484) GATTS_MAIN_C: Camera Init Failed␛[0m

closed 2020-07-18 by raphaelbs