raphaelbs/esp32-cam-ai-thinker #10: Stack smashing protection mode

amckay1, 2019-04-30:
Hi Raphael,

This is so very helpful, thank you! 

I hit a potential bug in examples/sd_jpg for which I think I found a partial fix. The default configuration for the ESP-IDF has stack smashing protection mode set to "Overall" which the example fails repeatedly (triggering restarts). Setting this to "Normal" in `make menuconfig` seems to fix the problem.

I'll try to learn more and hopefully find a fix that passes the more stringent protection mode.

Thanks again!

chrisMa68, 2019-05-19:
I had the same Problem, but after set 
CONFIG_STACK_CHECK_NORM=y
is works fine...

raphaelbs, 2019-05-20:
Can you share a piece of the bug? Also feel free to open a PR to add this configurations to the doc.