raphaelbs/esp32-cam-ai-thinker #13: Error in Http Client

ashishbharadwaj100, 2019-05-06:
Hi Raphaelbs,

I am your great fan from India.

Raphaelbs, You have done a mind-blowing job.

Please help me on this issue.

I (266759) example_google_storage: Taking picture...
I (267049) example_google_storage: Picture 2019-05-06_09-38-17.jpg taken.
I (267049) example_google_storage: Uploading 2019-05-06_09-38-17.jpg...
I (267059) example_google_storage: Insert object [2019-05-06_09-38-17.jpg] in bucket [ageyebucket]...
**E (270619) HTTP_CLIENT: This authentication method is not supported: Bearer realm="https://accounts.google.com/", error=invalid_token**
I (270619) example_google_storage: Image uploaded!
I (270619) example_google_storage: Status = 401
Heap summary for capabilities 0x00000004:
  At 0x3ffbcf8c len 32767 free 20268 allocated 12452 min_free 0
    largest_free_block 20268 alloc_blocks 2 free_blocks 1 total_blocks 3
  At 0x3f800000 len 4194303 free 3950772 allocated 243440 min_free 3932676
    largest_free_block 3945676 alloc_blocks 10 free_blocks 4 total_blocks 14
  At 0x3ffae6e0 len 6432 free 0 allocated 6292 min_free 0
    largest_free_block 0 alloc_blocks 27 free_blocks 0 total_blocks 27
  At 0x3ffb9610 len 158192 free 2772 allocated 154596 min_free 0
    largest_free_block 668 alloc_blocks 185 free_blocks 13 total_blocks 198
  At 0x3ffe0440 len 15072 free 16 allocated 14948 min_free 0
    largest_free_block 16 alloc_blocks 18 free_blocks 1 total_blocks 19
  At 0x3ffe4350 len 113840 free 102100 allocated 10360 min_free 70912
    largest_free_block 75776 alloc_blocks 300 free_blocks 37 total_blocks 337
  Totals:
    free 4076056 allocated 441968 min_free 4003588 largest_free_block 3945676

raphaelbs, 2019-05-20:
@ashishbharadwaj100 change log level to debug. This will make log more verbose and you can check whether your credentials are valid or not.

raphaelbs, 2020-07-19:
@ashishbharadwaj100 I'd update the documentation and added PlatformIO on this example. Also, I did re-tested and it is working fine.

Feel free to re-open this issue in case it is still going on.

closed 2020-07-19 by raphaelbs