raphaelbs/esp32-cam-ai-thinker #15: Error when loading in M5Cam

martarho, 2019-05-31:
I am trying to adapt the code in this repo to my M5Cam, but despite trying to force it to recognise the 4Mb RAM, it still crashes with this error:

E (489) spiram: Expected 8192KiB chip but found 4096KiB chip. Bailing out..

Are there any reasons why the code here should not work with my ESP32-based board? Thanks!

closed 2019-05-31 by martarho

raphaelbs, 2019-06-03:
You can try change `CONFIG_SPIRAM_TYPE` config in your `sdkconfig` as shown [in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html?highlight=sdkconfig#config-spiram-type).

martarho, 2019-06-04:
Hey, managed to fix it :) thanks

raphaelbs, 2019-06-04:
@martaho I would appreciate if you share what was the solution. Thanks