raphaelbs/esp32-cam-ai-thinker #18: OV7725 camera support

ashishbindal93, 2019-08-25:
#feature request

Please share code for OV7725 with ai-thinker esp32 board

[image]

raphaelbs, 2019-09-11:
Hello @ashishbindal93 
The code in this repository was heavily based in `https://github.com/espressif/esp32-camera`.
You can put together this examples with `OV7725` sensor support available there.

Also, I would be grateful with you contribute to this repo by uploading your code once you got it working.

ashishbindal93, 2019-09-11:
Didn't get that working. I have tried all the codes available but no code works for this setup.

ashishbindal93, 2019-09-11:
I can provide you pinning for the ov7725 camera module and esp cam board.

ashishbindal93, 2020-04-20:
Any update over OV7725?
Anyone got that working?

closed 2020-07-18 by raphaelbs