raphaelbs/esp32-cam-ai-thinker #20: SD stored pictures get by browser

PirSoft, 2019-09-18:
Hi Dears;
I would like stored sd card jpg files get by browser. capture triger with GPIO similar digital camera. one push button capture image1.jpg second push capture image2.jpg .
after stored SD card i would like access with browser, get files by link
for example : http://ipaddress/jpeg/image1.jpg enter browser and ge file from sd card.

how can I do it_?

thanks. Best Regards

raphaelbs, 2019-09-26:
That is an interesting project idea.
I believe that an example that takes a photo after pressing a button would help.

You can change the [sd example project](https://github.com/raphaelbs/esp32-cam-ai-thinker/tree/master/examples/sd_jpg) to trigger a new capture using the GPIO button.

PirSoft, 2019-10-01:
hi 

> That is an interesting project idea.
> I believe that an example that takes a photo after pressing a button would help.
> 
> You can change the [sd example project](https://github.com/raphaelbs/esp32-cam-ai-thinker/tree/master/examples/sd_jpg) to trigger a new capture using the GPIO button.

hi Raphaelbs;

I get image from sd car by web browser. but get onle small part of image. code at bellow. what is problem?

`

 File f = SD_MMC.open("/capture.jpg", "r");
  int filesize = f.size();
  String WebString = "";
  WebString += "HTTP/1.1 200 OK\r\n";
  WebString += "Content-Type: image/jpeg\r\n";
  WebString += "Content-Length: " + String(filesize) + "\r\n";
  WebString += "\r\n";
 
  char buf[1024];
  int siz = f.size();
  while(siz > 0) {
   size_t len = std::min((int)(sizeof(buf) - 1), siz);
    f.read((uint8_t *)buf, len);
    httpd_resp_set_type(req, "image/jpeg");
    httpd_resp_send(req,(const char*)buf,len);
    siz -= len; 
    f.close();
  } 
`