raphaelbs/esp32-cam-ai-thinker #24: CORRUPT HEAP - Google_Storage

Kuroskii, 2019-11-10:
@raphaelbs, estava tentando upar o seu código para a minha câmera, e me deparei com esse erro:

```
I (5832) event: sta ip: 10.0.0.120, mask: 255.255.255.0, gw: 10.0.0.1
I (5832) example_google_storage: SYSTEM_EVENT_STA_GOT_IP
I (5832) example_google_storage: Got IP: '10.0.0.120'
I (5842) example_google_storage: Initializing SNTP
I (5852) example_google_storage: Waiting for system time to be set... (1/10)
I (7852) example_google_storage: Date from NTP: 2019-11-10_16-51-41
I (7852) example_google_storage: Set SYNC_NTP
Heap summary for capabilities 0x00000004:
  At 0x3ffbd204 len 32767 free 15980 allocated 16740 min_free 15980
    largest_free_block 15980 alloc_blocks 2 free_blocks 1 total_blocks 3
  At 0x3f800000 len 4194303 free 3950904 allocated 243316 min_free 3930984
    largest_free_block 3949520 alloc_blocks 9 free_blocks 3 total_blocks 12
  At 0x3ffae6e0 len 6432 free 0 allocated 6292 min_free 0
    largest_free_block 0 alloc_blocks 27 free_blocks 0 total_blocks 27
  At 0x3ffb9888 len 157560 free 4096 allocated 152824 min_free 96
    largest_free_block 4008 alloc_blocks 147 free_blocks 5 total_blocks 152
  At 0x3ffe0440 len 15072 free 320 allocated 14676 min_free 148
    largest_free_block 320 alloc_blocks 10 free_blocks 1 total_blocks 11
  At 0x3ffe4350 len 113840 free 110588 allocated 3208 min_free 108680
    largest_free_block 110588 alloc_blocks 2 free_blocks 1 total_blocks 3
  Totals:
    free 4081888 allocated 437056 min_free 4055888 largest_free_block 3949520
I (27932) example_google_storage: Refresh auth token...
CORRUPT HEAP: multi_heap.c:432 detected at 0x78447664
abort() was called at PC 0x40098a43 on core 0
0x40098a43: multi_heap_assert at C:/msys32/home/Acer/esp/esp-idf/components/heap/multi_heap.c:716
 (inlined by) multi_heap_malloc_impl at C:/msys32/home/Acer/esp/esp-idf/components/heap/multi_heap.c:432

ELF file SHA256: 93ed5a714f7849f63954918d2dc9cd7cfd6baf181db015b6f484f1faf0c62bec

Backtrace: 0x4008ca93:0x3ffc0ed0 0x4008cd88:0x3ffc0f00 0x40098a43:0x3ffc0f30 0x40085b34:0x3ffc0f60 0x40085ba1:0x3ffc0f90 0x400865b9:0x3ffc0fc0 0x4000beaf:0x3ffc0ff0 0x40086654:0x3ffc1010 0x400
0bef5:0x3ffc1040 0x401542e0:0x3ffc1060 0x4012c91d:0x3ffc1090 0x400d7473:0x3ffc1110 0x40095e42:0x3ffc1180
0x4008ca93: invoke_abort at C:/msys32/home/Acer/esp/esp-idf/components/esp32/panic.c:716

0x4008cd88: abort at C:/msys32/home/Acer/esp/esp-idf/components/esp32/panic.c:716

0x40098a43: multi_heap_assert at C:/msys32/home/Acer/esp/esp-idf/components/heap/multi_heap.c:716
 (inlined by) multi_heap_malloc_impl at C:/msys32/home/Acer/esp/esp-idf/components/heap/multi_heap.c:432

0x40085b34: heap_caps_malloc at C:/msys32/home/Acer/esp/esp-idf/components/heap/heap_caps.c:354

0x40085ba1: heap_caps_malloc_default at C:/msys32/home/Acer/esp/esp-idf/components/heap/heap_caps.c:354

0x400865b9: _malloc_r at C:/msys32/home/Acer/esp/esp-idf/components/newlib/syscalls.c:37

0x40086654: _calloc_r at C:/msys32/home/Acer/esp/esp-idf/components/newlib/syscalls.c:58

0x401542e0: esp_http_client_init at C:/msys32/home/Acer/esp/esp-idf/components/esp_http_client/esp_http_client.c:1236

0x4012c91d: gcp_auth_refresh_token at C:/msys32/home/Acer/esp/esp32-cam-ai-thinker/examples/google_storage/components/google_api/gcp_auth_refresh_token.c:47

0x400d7473: task_gcp at C:/msys32/home/Acer/esp/esp32-cam-ai-thinker/examples/google_storage/main/main.c:122

0x40095e42: vPortTaskWrapper at C:/msys32/home/Acer/esp/esp-idf/components/freertos/port.c:436
```
Aparentemente, quando ele chama o `gcp_auth_refresh_token()` em `task_gcp`, ele tenta alocar em um espaço de memória já ocupado. Tem ideia do que possa ser/como resolver? estou usando a ESP-IDF stable version.
ele puxa bibliotecas da pasta raiz `esp/esp-idf/components`, é normal mesmo? ou devo puxar essas bibliotecas para o repositório `google_storage`?

raphaelbs, 2019-11-11:
Olá @Kuroskii 
Sobre o erro de memória, é possível que suas crendenciais sejam maiores que o espaço disponível.
Dê uma olhada [aqui](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/examples/google_storage/components/google_api/gcp_auth_refresh_token.c#L38).

Não sou expert em C++. Fiz esse código (`malloc` + `sprintf`) com intuito de concatenar as strings.
Se você conhecer uma alternativa melhor, fique a vontade para abrir um PR.

Sobre as referências das bibliotecas, as definições das origens das bibliotecas ficam no [CMakeLists](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/examples/google_storage/CMakeLists.txt).

Kuroskii, 2019-11-11:
Obrigado, durante a semana é meio difícil de conseguir mexer, mas no fim de semana te dou um retorno.

Kuroskii, 2019-11-16:
Deu certinho! Agora o próximo passo é pegar informação do realtime database do firebase, pra fazer o upload das fotos apenas quando a booleana tiver true.

raphaelbs, 2019-11-16:
Bom saber!
Se puder, abra um PR com o exemplo de leitura do Realtime database.

closed 2019-11-16 by raphaelbs