raphaelbs/esp32-cam-ai-thinker #26: Boot error: SOC_RESERVE_MEMORY_REGION region range

imShara, 2019-11-16:
Binary built with last esp-idf. When trying to run its throws error and reboots

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7228
load:0x40078000,len:15568
load:0x40080400,len:4772
entry 0x400806f0
I (74) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.1-dev-997-gd57890cdf 2nd stage bootloader
I (39) boot: compile time 00:42:06
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x23538 (144696) map
I (164) esp_image: segment 1: paddr=0x00033560 vaddr=0x3ffb0000 size=0x0361c ( 13852) load
I (170) esp_image: segment 2: paddr=0x00036b84 vaddr=0x40080000 size=0x00400 (  1024) load
I (171) esp_image: segment 3: paddr=0x00036f8c vaddr=0x40080400 size=0x0908c ( 37004) load
I (195) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x93584 (603524) map
I (414) esp_image: segment 5: paddr=0x000d35ac vaddr=0x4008948c size=0x11070 ( 69744) load
I (460) boot: Loaded app from partition at offset 0x10000
I (460) boot: Disabling RNG early entropy source...
I (460) psram: This chip is ESP32-D0WD
I (466) spiram: Found 64MBit SPI RAM device
I (469) spiram: SPI RAM mode: flash 40m sram 40m
I (475) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (482) cpu_start: Pro cpu up.
I (486) cpu_start: Application information:
I (490) cpu_start: Project name:     esp32_cam_http_jpg
I (496) cpu_start: App version:      88a99fb
I (501) cpu_start: Compile time:     Nov 17 2019 00:41:54
I (507) cpu_start: ELF file SHA256:  b5d9ea31f9c8df36...
I (513) cpu_start: ESP-IDF:          v4.1-dev-997-gd57890cdf
I (520) cpu_start: Starting app cpu, entry point is 0x400815bc
I (507) cpu_start: App cpu up.
I (1415) spiram: SPI SRAM memory test OK
D (1416) memory_layout: Checking 8 reserved memory ranges:
D (1416) memory_layout: Reserved memory range 0x3f800000 - 0x40000000
E (1421) memory_layout: SOC_RESERVE_MEMORY_REGION region range 0x3f800000 - 0x40000000 overlaps with 0x3ffae000 - 0x3ffae6e0
abort() was called at PC 0x400eab1b on core 0

ELF file SHA256: b5d9ea31f9c8df3659652b4b99ec3a8664b7e056aae8481852fa4b9fdf9235a9
                                                                                                             
Backtrace: 0x40090879:0x3ffe3450 0x40090cec:0x3ffe3480 0x400eab1b:0x3ffe34b0 0x400eabc8:0x3ffe34f0 0x400d6b5a:0x3ffe3840 0x40081821:0x3ffe3bf0 0x400796a3:0x3ffe3c30 |<-CORRUPTED
```

imShara, 2019-11-19:
Just made working example from scratch https://github.com/imShara/ESP32-CAM-XRZ00D1-V240-HTTP-AP

avarpow, 2019-11-24:
I have the same problem as you.Have you solved the problem？
```Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7208
load:0x40078000,len:15568
ho 0 tail 12 room 4
load:0x40080400,len:4772
entry 0x400806f0
I (74) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.1-dev-1086-g93a8603c5 2nd stage bootloader
I (41) boot: compile time 20:07:14
I (41) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 2MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x244f4 (148724) map
I (167) esp_image: segment 1: paddr=0x0003451c vaddr=0x3ffb0000 size=0x03690 ( 13968) load
I (173) esp_image: segment 2: paddr=0x00037bb4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/8192/Desktop/feibiao/esp-idf/components/freertos/xtensa_vectors.S:1778

I (174) esp_image: segment 3: paddr=0x00037fbc vaddr=0x40080400 size=0x0805c ( 32860) load
I (197) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x917dc (595932) map
0x400d0020: _stext at ??:?

I (413) esp_image: segment 5: paddr=0x000d1804 vaddr=0x4008845c size=0x1675c ( 91996) load
0x4008845c: esp_flash_read at C:/Users/8192/Desktop/feibiao/esp-idf/components/spi_flash/esp_flash_api.c:493

I (470) boot: Loaded app from partition at offset 0x10000
I (471) boot: Disabling RNG early entropy source...
I (471) psram: This chip is ESP32-D0WD
I (477) spiram: Found 64MBit SPI RAM device
I (480) spiram: SPI RAM mode: flash 40m sram 40m
I (485) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (493) cpu_start: Pro cpu up.
I (496) cpu_start: Application information:
I (501) cpu_start: Project name:     esp32_cam_http_jpg
I (507) cpu_start: App version:      88a99fb
I (512) cpu_start: Compile time:     Nov 24 2019 13:49:30
I (518) cpu_start: ELF file SHA256:  71d401960418e61d...
I (524) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5
I (531) cpu_start: Starting app cpu, entry point is 0x400815cc
0x400815cc: call_start_cpu1 at C:/Users/8192/Desktop/feibiao/esp-idf/components/esp32/cpu_start.c:276

I (517) cpu_start: App cpu up.
I (1426) spiram: SPI SRAM memory test OK
V (1427) memory_layout: reserved range is 0x3f4244e4 - 0x3f424514
D (1427) memory_layout: Checking 8 reserved memory ranges:
D (1431) memory_layout: Reserved memory range 0x3f800000 - 0x40000000
E (1438) memory_layout: SOC_RESERVE_MEMORY_REGION region range 0x3f800000 - 0x40000000 overlaps with 0x3ffae000 - 0x3ffae6e0
abort() was called at PC 0x400eb70f on core 0
0x400eb70f: s_prepare_reserved_regions at C:/Users/8192/Desktop/feibiao/esp-idf/components/soc/src/memory_layout_utils.c:109 (discriminator 1)

ELF file SHA256: 71d401960418e61d61ace4fb6f16c2910cc8446f7866059a5f94c69c50a147b4

Backtrace: 0x40094f35:0x3ffe3430 0x400953a8:0x3ffe3460 0x400eb70f:0x3ffe3490 0x400eb7d1:0x3ffe34d0 0x400d6d56:0x3ffe3840 0x40081831:0x3ffe3bf0 0x400796a3:0x3ffe3c30 |<-CORRUPTED
0x40094f35: invoke_abort at C:/Users/8192/Desktop/feibiao/esp-idf/components/esp32/panic.c:157

0x400953a8: abort at C:/Users/8192/Desktop/feibiao/esp-idf/components/esp32/panic.c:174

0x400eb70f: s_prepare_reserved_regions at C:/Users/8192/Desktop/feibiao/esp-idf/components/soc/src/memory_layout_utils.c:109 (discriminator 1)

0x400eb7d1: soc_get_available_memory_regions at C:/Users/8192/Desktop/feibiao/esp-idf/components/soc/src/memory_layout_utils.c:126

0x400d6d56: heap_caps_init at C:/Users/8192/Desktop/feibiao/esp-idf/components/heap/heap_caps_init.c:67

0x40081831: call_start_cpu0 at C:/Users/8192/Desktop/feibiao/esp-idf/components/esp32/cpu_start.c:261

```

imShara, 2019-11-24:
@avarpow this example out of date I think. I built new from scratch, [check it](https://github.com/imShara/ESP32-CAM-XRZ00D1-V240-HTTP-AP)

avarpow, 2019-11-24:
@imShara It works!Thanks for your tips.

raphaelbs, 2019-12-10:
@imShara can you help me updating this repo?

imShara, 2019-12-11:
@raphaelbs will try, but not in the near future..
Try to compare initialization with my version of http_jpg

raphaelbs, 2019-12-11:
Hey @imShara,
I appreciate that, thanks!

wero1414, 2020-03-04:
Hi @imShara Im really happy to find this! you really help me out!

raphaelbs, 2020-07-19:
I'd rebuilt the examples with PlatformIO.
Feel free to re-open the issue in case it is still happen.

closed 2020-07-19 by raphaelbs

Tomas1337, 2021-04-11:
Hmmm. Still happening with PlatformIO with me