raphaelbs/esp32-cam-ai-thinker #27: mbedtls_ssl_handshake returned -0x10

Kirankumar-xenlabs, 2019-11-21:
## Environment

- Development Kit: ESP32-S AI-Thinker
- Kit version (for ESP32-S AI-Thinke): none
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v3.2.2
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a5
- Operating System: Windows
- Power Supply: USB

## Problem Description
I am using google storage example, It uploading up to 24 images on firebase, after that it fails to open the new connection and mbedtls_ssl_handshake returned -0x10.

### Code to reproduce this issue

```
esp_err_t gcp_storage_insert_object(const char *pic_name, const char *binary, size_t binary_size)
{
  ESP_LOGI(TAG, "Insert object [%s] in bucket [%s]...", pic_name, CONFIG_GCP_BUCKET);

  const char *URL = "https://www.googleapis.com/upload/storage/v1/b/%s/o?uploadType=media&name=%s";
  char *post_url = malloc(strlen(URL) + strlen(CONFIG_GCP_BUCKET) + strlen(pic_name));
  sprintf(post_url, URL, CONFIG_GCP_BUCKET, pic_name);

  esp_http_client_config_t config = {
      .url = post_url,
      .event_handler = gcp_build_event_handle(http_handler_cb),
      .method = HTTP_METHOD_POST};
  esp_http_client_handle_t http_client = esp_http_client_init(&config);
  esp_http_client_set_post_field(http_client, binary, binary_size);

  esp_http_client_set_header(http_client, "Content-Type", "image/jpg");
  esp_http_client_set_header(http_client, "Authorization", ACCESS_TOKEN);

  esp_err_t err = esp_http_client_perform(http_client);
  if (err == ESP_OK)
  {
    ESP_LOGI(TAG, "Status = %d", esp_http_client_get_status_code(http_client));
  }
  esp_http_client_cleanup(http_client);

  free(post_url);

  return http_status;
} 
```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6200
load:0x40078000,len:11208
load:0x40080400,len:7124
entry 0x40080764
I (28) boot: ESP-IDF v3.2.2 2nd stage bootloader
I (28) boot: compile time 11:03:56
I (29) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
[... 675 lines cut ...]
E (151394) esp-tls: mbedtls_ssl_handshake returned -0x4d80
E (151404) esp-tls: Failed to open new connection
E (151404) TRANS_SSL: Failed to open a new connection
E (151404) HTTP_CLIENT: Connection failed, sock < 0

After Upload To GCS DRAM 4189992 IRAM 4197300
DRAM 4189992 IRAM 4197300

I (151414) example_google_storage: After Upload -----> free DRAM 4190056 IRAM 4197364
I (151424) example_google_storage: After memory release -----> free DRAM 4190056 IRAM 4197364

Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
Waiting for bell trigger....
```

## Other items if possible

- [sdkconfig.txt](https://github.com/raphaelbs/esp32-cam-ai-thinker/files/3873915/sdkconfig.txt)

raphaelbs, 2019-12-10:
Hello @Kirankumar-xenlabs 
Sorry about the delay. Have you managed to figure it out?

Kirankumar-xenlabs, 2019-12-11:
Hello @raphaelbs, Thanks for the reply, I fixed it

closed 2019-12-11 by Kirankumar-xenlabs