raphaelbs/esp32-cam-ai-thinker #29: does not compile with rtc_gpio_desc, should be change to rtc_io_desc in twi.c

julienfr112, 2020-01-18:


raphaelbs, 2020-01-22:
Hi @julienfr112,
Can you provide extra information about the problem?

aquaticus, 2020-01-23:
@julienfr112 You must enable option `RTCIO_SUPPORT_RTC_GPIO_DESC` to get rid of this compilation error.
Run `menuconfig` and go to `Component config → Driver configurations → RTCIO configuration` and check `Support array 'rtc_gpio_desc' for ESP32`

bill-orange, 2020-09-18:
I am somewhat new to VSCODE and I am unable to follow your instructions.  Where is menuconfig and how do you run it?

raphaelbs, 2020-09-18:
Hey @bill-orange, how are you doing?

I recommend not using `make` to build the project. PlatformIO is much easier (or at least, it should be).

As soon as you open the project on VSCode, it should prompt you to install the PlatformIO extension (if you haven't already). Then, select the folder that you are interested to build and hit the checkmark button on the bottom left corner. The [GIF](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/assets/using-with-platformio.gif) shows this steps.

Regarding the config file, every example has a `Kconfig.projbuild` file. You can set the default values into it. Note that when you hit the build button on VSCode, it will generate a `sdkconfig` file that contains all the environment variables that the build step will use. In case you need to change some values, you can directly change on the file and re-build again or you can delete it and re-build again.

I noticed that there is some documentation indicating the `make menuconfig`. That should not be there anymore.

bill-orange, 2020-09-18:
I am sure you put me on the right track.  However, I am not there yet.  I added `CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC=y` into sdkconfig.defaults and still no joy.  Same error. I also tried adding the subject line to sdkconfig with the same negative result.

I would have trued Kconfig.projbuild but I could not figure out the syntax, what to put where.

Thoughts?

bill-orange, 2020-09-19:
I changed to` rtc_io_desc` and cam compile successfully.  I am not sure if the problem is solved so I will leave this open.

raphaelbs, 2020-09-20:
Glad you got it working @bill-orange!

closed 2020-09-20 by raphaelbs

bill-orange, 2020-09-20:
Only a partial victory. When it runs I get a memory conflict. I opened another issues since this one is resolved.