raphaelbs/esp32-cam-ai-thinker #32: Reboot and flash through programmer

luzat, 2020-02-29:
To ease development with my FTDI FT232R interface I found that the following connections are useful:

ESP32 Pin | Serial Pin | Comment
-|-|-
EN | RTS | `EN` = `CHIP_PU` (I think). Not broken out, but reachable through the soldered connection of the board button which points inward. Can be used to temporarily attach/solder a cable.
GPIO0 | DTR | Labelled pin on the board

This can be done instead of shorting `GPIO0` and `GND` and hitting reset. The tools automatically trigger a reboot and flashing mode that way. This should also work with CP2102 boards which have `DTR` and `RTS` available.

Additional information regarding the connection: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader The `CHIP_PU` connection was deduced from the schematic.

Might be helpful to add that information?

luzat, 2020-02-29:
Also, possibly include it in the wiring diagram.

raphaelbs, 2020-03-04:
Oh wow! That's really helpful!

@luzat feel free to open a pull request adding this into the [Instructions](https://github.com/raphaelbs/esp32-cam-ai-thinker#instructions) section.