raphaelbs/esp32-cam-ai-thinker #34: Fail on motion detect (change_detect example)

thomas-pegot, 2020-03-22:
Hi,
I flashed the change_detect example and it seemed to failed on during motion detection 👍 

`D (4487) camera: Allocating DMA buffer #0, size=1280
D (4497) camera: Allocating DMA buffer #1, size=1280
D (4507) camera: Allocating DMA buffer #2, size=1280
D (4507) camera: Allocating DMA buffer #3, size=1280
I (4517) camera: Allocating 2 frame buffers (37 KB total)
I (4517) camera: Allocating 18 KB frame buffer in OnBoard RAM
I (4527) camera: Allocating 18 KB frame buffer in OnBoard RAM
E (4527) gpio: gpio_install_isr_service(409): GPIO isr service already installed
E (4537) camera: gpio_install_isr_service failed (103)
E (4547) camera: Camera init failed with error 0x103
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40089ce8
0x40089ce8: _esp_error_check_failed at C:/Users/mrtom/OneDrive/Bureau/esp-idf-3/components/esp32/panic.c:726

file: "../main/main.c" line 139
func: change_detector
expression: esp_camera_init(&detection_config)

ELF file SHA256: 1021f35fa30c9879509aa9f14268eb33826956a052f05b14028a7b07a30fc08e

Backtrace: 0x40089589:0x3ffc12f0 0x40089ceb:0x3ffc1320 0x400d892f:0x3ffc1350 0x4008cf5a:0x3ffc1390
0x40089589: invoke_abort at C:/Users/mrtom/OneDrive/Bureau/esp-idf-3/components/esp32/panic.c:155

0x40089ceb: _esp_error_check_failed at C:/Users/mrtom/OneDrive/Bureau/esp-idf-3/components/esp32/panic.c:727

0x400d892f: change_detector at c:\users\mrtom\onedrive\bureau\esp-idf-3\esp32-cam-ai-thinker\examples\change_detection\build/../main/main.c:139 (discriminator 1)

0x4008cf5a: vPortTaskWrapper at C:/Users/mrtom/OneDrive/Bureau/esp-idf-3/components/freertos/port.c:143`

I wonder if it's not due to the fact that esp_camera_init can only be called once.

thomas-pegot, 2020-03-22:
```
/**
 * @brief Initialize the camera driver
 *
 * @note call camera_probe before calling this function
 *
 * This function detects and configures camera over I2C interface,
 * allocates framebuffer and DMA buffers,
 * initializes parallel I2S input, and sets up DMA descriptors.
 *
 * Currently this function can only be called once and there is
 * no way to de-initialize this module.
 *
 * @param config  Camera configuration parameters
 *
 * @return ESP_OK on success
 */
esp_err_t esp_camera_init(const camera_config_t* config);
```

So it seems logic that this trick shouldn't work;

```
esp_camera_deinit();
ESP_ERROR_CHECK(esp_camera_init(&photo_config));
```

raphaelbs, 2020-04-15:
Hey @thomas-pegot, thanks for reporting this issue!

Looking at the [deinit function](https://github.com/espressif/esp32-camera/blob/master/driver/camera.c#L1361), it does remove the GPIO handler but this doesn't seems to be enough.
Can you try calling [`gpio_uninstall_isr_service`](https://github.com/espressif/esp-idf/blob/d6728090801a95292dbaa343b508f284f0ef0736/components/driver/gpio.c#L481) after the [`esp_camera_deinit`](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/examples/change_detection/main/main.c#L158)?

thomas-pegot, 2020-04-18:
Hi @raphaelbs ,
I tried it and I had the same outcome.
So it happens when it calls [gpio_install_isr_service](https://github.com/espressif/esp32-camera/commit/ec274179ebc16fd85f6f785f6aa7d8a794766b4b) it changed on 29dec 2019.
We can try/catch this error  409 to bypass it. But since it's a "goto fail" idk if we can fo much.
Honestly goto fail is pretty ugly.

raphaelbs, 2020-04-18:
Oh I see. That explains why it was working before!

Using the uninstall method should've worked as it [sets the `gpio_isr_func` to `NULL`](https://github.com/espressif/esp-idf/blob/d6728090801a95292dbaa343b508f284f0ef0736/components/driver/gpio.c#L489) and the install method [only asserts the value of `gpio_isr_func`](https://github.com/espressif/esp-idf/blob/d6728090801a95292dbaa343b508f284f0ef0736/components/driver/gpio.c#L438).

I guess we should investigate other methods of re-config the camera.

raphaelbs, 2020-04-18:
The self-hosted camera driver has the outdated version:
https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/components/ov2640/camera/camera.c#L1165

The build should work with it.

thomas-pegot, 2020-04-19:
I mean what are the pros to reset camera ?
This code can work fine without:
` esp_camera_deinit();
ESP_ERROR_CHECK(esp_camera_init(&photo_config));`

and add `esp_camera_fb_return(fb)` at the end of the loop.

Could for example make a different branch to denote version with old esp_camera lib and one with new esp_camera that stay updated.
Just some thought and thank you for your reply.

closed 2020-04-19 by thomas-pegot