raphaelbs/esp32-cam-ai-thinker #51: Failure to compile with memory error

bill-orange, 2020-09-19:
I am trying to run the change detection example.  After much consternation I got it to compile in platformIO.  My attemps to run it have not been successful.  Here is the failure.

```
␛[0;32mI (1214) spiram: SPI SRAM memory test OK␛[0m
D (1215) memory_layout: Checking 8 reserved memory ranges:␛[0m
D (1215) memory_layout: Reserved memory range 0x3f800000 - 0x40000000␛[0m
␛[0;31mE (1220) memory_layout: SOC_RESERVE_MEMORY_REGION region range 0x3f800000 - 0x40000000 overlaps with 0x3ffae000 - 0x3ffae6e0␛[0m
abort() was called at PC 0x400dc81b on core 0

ELF file SHA256: 9af668d35db12fb1

Backtrace: 0x40089370:0x3ffe34a0 0x4008981c:0x3ffe34d0 0x400dc81b:0x3ffe3500 0x400dc8c0:0x3ffe3540 0x400db895:0x3ffe3890 0x40082599:0x3ffe3c40 0x4007931a:0x3ffe3c80 |<-CORRUPTED

Rebooting...
```

Any ideas on what could be wrong?

bill-orange, 2020-09-23:
This problem was resolved by adding the following fix to sdkconfig.  This would be needed on all examples.

```
#CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_TYPE=Auto-detect
```

Now that I can compile and run, I can observe ( via serial) motion being detected and higher resolution images being taken. Hurrah.
What I do not see in the code is the storage of the highres image or any way to access it.  I that left to the user or am I missing something?  Nothing is written to a FAT32 usd card nor do I see code to accomplish it.  I can certainly add the code but I want to be sure I understand what's happening as-is.

Thoughts?

raphaelbs, 2020-12-15:
hey @bill-orange, thanks for pointing the `CONFIG_SPIRAM_TYPE` flag out. I'll update it.

Regarding storing the high res picture on the SD, this part isn't implemented on the change detection example.

Vitenuto, 2022-08-23:
Man I love you, finally after several hours/days of searching how to run it I was able to run the google_storage example! 
Thank you so much!